lsellens/octoprint-manual_bed_tramming #1: Use getAdditionalControls callback in frontend instead of manipulating the config in the backend

foosel, 2018-02-26:
FYI, instead of injecting your controls into the configuration:

https://github.com/electr0sheep/OctoPrint-Cr10_leveling/blob/4942918fe10b886a371685b78e96289e9167869a/octoprint_CR10_Leveling/__init__.py#L11-L22

you should declare your own JS asset with a view model and use the [`getAdditionalControls()` view model callback](http://docs.octoprint.org/en/master/plugins/viewmodels.html#callbacks) instead - it's what it was made to do and runs less risk of interfering with people's settings ;)

electr0sheep, 2018-02-26:
Thanks for the tip, I'll see what I can do!

electr0sheep, 2018-02-27:
Thanks again for the tip @foosel, this makes everything much smoother!

closed 2018-02-27 by electr0sheep