lsellens/octoprint-manual_bed_tramming #3: "Upper" left broken

bcsteeve, 2018-02-26:
Upper left and lower left go to the same place.  Right works fine.  Center does too.  Would be ideal if we could specify coords for each.

nean-and-i, 2018-02-26:
typo "F290" instead of "Y290" created a pull request

electr0sheep, 2018-02-27:
Resolved with f05e84c27449dedb3d2dcfec2c548e91d2abc4f6

electr0sheep, 2018-02-27:
Actually, before I resolve this I want to get more information

> Would be ideal if we could specify coords for each.

Since you are looking to specify the heat temperature as well as the corner coordinates, I am wondering if you are looking for a plugin like this:
https://github.com/Salandora/octoprint-customControl
Instead of this plugin. customControl can be used in conjunction with this to modify whatever commands you want.

The idea behind this plugin was to be easy-to-use. I can't imagine many people would want to configure the location of the corners. If I am wrong, please leave a comment, and I'll work on implementing customizable controls.

bcsteeve, 2018-02-27:
I'm happy you're receptive to feedback.  The ideal method for bed leveling is if the nozzle is directly over the adjustment screws.  Your default coordinates aren't over the screws.  I precisely position mine (eg 31.2, 27.7) and it is much quicker to level than I'd I'm just sorta close... but close is better than not close.  So even if you use default values that are "close enough", it would be better.  I have two CR-10's so I can tell you they are not consistently manufactured to a level where I can give you my precise figures and we expect then to work for everyone.  They didn't even space then evenly on the corners.  The front-left on one of mine is 3mm for in from the edge than the back left.  It is this variability of manufacturing that is the basis for my request that we customize.  However, if you want to keep it static, I suggest measuring yours (rounded to the nearest mm) and use those for defaults rather than 10, 10 - which is nowhere near the screws.

electr0sheep, 2018-02-27:
Great, with this in mind, you've convinced me that it would be a genuinely useful feature.

If you have those measurements on hand, and want to let me know in a comment, I can change the static positioning and release an update quickly.

Configurable controls will probably take me a little longer.

Also, I have released a new version 1.1.0 that should make installation/enabling/disabling/and uninstallation much smoother. However, it is no longer configurable by customControls.

Thank you for your feedback, and for trying out my plugin. I am new to CR-10s, OctoPrint, and OctoPrint plugin development and your feedback is most welcome!

bcsteeve, 2018-02-27:
Well, for me on the printer I'm using at the moment it is:

Front left:  x31.4 y27.1
Back left:  x30.2 y268.0
Back right: x271.1 y268.6
Front right: x270.7 y27.9

I would also suggest using that labeling since "upper" and "lower" is less clear (to me).

I'm going to try your 1.1.0.

I wish I knew more about the coding side to help you more.  That's why I was so happy to see this plugin because I've wanted something like this but have no idea how to do it.

nean-and-i, 2018-02-27:
indeed, "front" and "back" would be better than "upper" and "lower"

regarding the coordinates, well, i'd also recommend a position not to close to screws or the end of a bed, why, because it could cause issues with some sorts of mods such as fan ducts , also a calibration more inner would be more efficient as some beds tend to bend,...

I have a tevo tornado, but its an identical clone of the cr-10,...

bcsteeve, 2018-02-27:
I'm going to respectfully disagree with above.  I think "at the screws" is the one "best" position (there are mathematical reasons why but we don't have to get into it).  But... it raises a point that "different strokes for different folks" is going to be the ultimate answer.  If I were you, @electr0sheep, I would simply document how to edit the .js file for those of us that want what we want.  Now that you've gone to that .js method (great move, I think), it is really easy to do and a simple browser refresh solves the issue.  Before, it was a nightmare figuring out which file then having to delete another file, then having to do a full system reboot.... it was driving me nuts.  Now... easy.

nean-and-i, 2018-02-27:
@bcsteeve , indeed, seems we booth have different view on that, and I don't take it personally, no worries ;-) Please allow me to share some of my experienced, that might differ to yours.

That said, I got the very best calibration results, when printing couple of rectangles, from outside to inside, that becomes smaller. 
I have lots of arguments that are not based on theoretical or mathematical reasons eg. glas should be plane, and not bend, the z-axis phenomena (where the x-axis can have tension and causes oozing/blobbing caused by to to tighten bearing opposite the stepper) -> you can calibrate until you get insane, etc., etc.
 
We all know that those cr-10/tevo tornado models, with questionable quality, are far away from just applying mathematical concepts, and everything is fine.
In this area we'd to rely on best practices/experiences,...
one of it is: pls. not too close to the screws (pls. at min. the whole printerhead+fan duct away), else some would cry out loud because on the first corner the screw (that holds the bed down) has disassembled their fancy fan duct,....

It may well be, that you have fixed all kind of issues and you have built a high precision device out of a cheap printer, but pls. respect all the starters who have a looong path ahead to get there, so,  safety first,...

thanks

bcsteeve, 2018-02-27:
Starters shouldn't be jumping into crazy over-sized fan mounts :)

electr0sheep, 2018-02-28:
I'm going to close this now, I have added customizable controls (just in the process of figuring out why the settings menu is blank for the actual release)

closed 2018-02-28 by electr0sheep