lsellens/octoprint-manual_bed_tramming #4: "Apply Heat" shouldn't be your values

bcsteeve, 2018-02-26:
They should be my values, not some random figure you thought would be nice.  It defaults to 50/170.  Why not *at least* go to what user has set for PLA?  Better... let them choose a profile

nean-and-i, 2018-02-26:
this should be bed:80 and noozle:190, especially for leveling a higher bed temp is useful

bcsteeve, 2018-02-26:
It shouldn't be any arbitrary value.  It should be user settable

electr0sheep, 2018-02-27:
What would be the best way to implement the change? Text fields where users can enter their own values, or a drop down box where users can select a material with preset heats?

At the same time, I will say that the purpose of "Apply Heat" isn't to preheat the bed and nozzle to printing temperature. The purpose is so that you don't level the bed cold, just to have the nozzle and bed expand due to the heat and be unlevel when it has been heated to printing temperature.

bcsteeve, 2018-02-27:
I suggest... if it is possible (I don't know) that you tap into the code that is already there where the user defined temperature/material profiles and then show a dropdown.  Bed leveling is ideally performed with the same influencing factors as the printing.  So it should be the same temp the print job will use.

If it isn't worthwhile making it a UI element, then just give some details on how to edit the file and make the changes stick (you gave me a hint on the uninstall issue).  Perhaps I'm more precise than most and not enough people care? I'm ok with that.

thisiskeithb, 2018-02-27:
Pulling from the system values for preheat settings would be nice, but I'm not sure what Octoprint exposes to plugins.

electr0sheep, 2018-02-28:
What I'll probably end up doing for the first attempt is to have a number input for the values, and then I can try to populate a drop down based on the system values. I'm pretty sure those settings are exposed. I pretty much have all that done, but the controls aren't updated until the user refreshes the page, so that's what's going on with that.

electr0sheep, 2018-02-28:
#9

closed 2018-02-28 by electr0sheep