lsellens/octoprint-manual_bed_tramming #7: Can't uninstall???

bcsteeve, 2018-02-27:
Because it wasn't working well, I disabled it... still there.  I removed it... still there.  I restarted.  Rebooted.  Still there!!  I ssh'd into the system and confirmed the files do not exist.... STILL THERE.

I cleared my cookies.  Different browser.   STILL THERE.

How is this even possible?

How in the world do I get it to go away?

electr0sheep, 2018-02-27:
I will work on making it easier to uninstall. If you ssh into your system, open the file ~/.octoprint/config.yaml and you'll see the added controls under controls:

If you remove the controls from there, it will get rid of them. Not the most convenient thing to do, admittedly.

bcsteeve, 2018-02-27:
Thanks!  I had no idea why edits weren't taking, and neither was removal.  Now I get it.  Not ideal, but functional... So thank you

bcsteeve, 2018-02-27:
Sorry... still not where I need to be.  I edited the yaml file and reloaded the page.  Still there.  Is restart/reboot/browser cache/cookie flush required?

electr0sheep, 2018-02-27:
Yeah, restart OctoPrint, if it's still there do a hard refresh, and if it's still there let me know.

Sorry about the trouble, you're kind of my guinea pig!

ryanneve, 2018-02-28:
I had the same problem, but it isn't obvious to me exactly which lines I should remove from config.yaml

```
accessControl:
  salt: LCx0ofT7Jjq3co89ki2t7AeO2AZxO045
api:
  key: B13DEBC3EDA649ED8CBF6E5696003E82
controls:
- children:
  - commands:
    - M140 S50
    - M104 S170
    - M190 S50
    - M109 S170
    - ''
    - ; Play tune
    - M300 S0 P66
    - M300 S987 P66
    - M300 S0 P66
    - M300 S1975 P66
    - M300 S0 P66
    - M300 S2959 P66
    - M300 S0 P66
    - M300 S2489 P66
    - M300 S0 P66
    - M300 S1975 P66
    - M300 S2959 P66
    - M300 S0 P133
    - M300 S2489 P133
    - M300 S0 P133
    - M300 S2093 P66
    - M300 S0 P66
    - M300 S4186 P66
    - M300 S0 P66
    - M300 S3135 P66
    - M300 S0 P66
    - M300 S2637 P66
    - M300 S0 P66
    - M300 S4186 P66
    - M300 S3135 P66
    - M300 S0 P133
    - M300 S2637 P133
    - M300 S0 P133
    - M300 S987 P66
    - M300 S0 P66
    - M300 S1975 P66
    - M300 S0 P66
    - M300 S2959 P66
    - M300 S0 P66
    - M300 S2489 P66
    - M300 S0 P66
    - M300 S1975 P66
    - M300 S2959 P66
    - M300 S0 P133
    - M300 S2489 P133
    - M300 S0 P133
    - M300 S2489 P66
    - M300 S2637 P66
    - M300 S2793 P66
    - M300 S0 P66
    - M300 S2793 P66
    - M300 S2959 P66
    - M300 S3135 P66
    - M300 S0 P66
    - M300 S3135 P66
    - M300 S3322 P66
    - M300 S1760 P66
    - M300 S0 P66
    - M300 S1975 P100
    customClass: btn btn-danger
    name: Apply Heat
    offset: '2'
    width: '11'
  - commands:
    - G1 Z10 F500
    - G1 X10 Y10 F3600
    - G1 Z0 F500
    - ''
    - ''
    customClass: btn
    name: Upper Left
    width: '2'
  - commands:
    - G1 Z10 F500
    - G1 X290 Y10 F3600
    - G1 Z0 F500
    - ''
    customClass: btn
    name: Upper Right
    offset: '2'
    width: '2'
  - commands:
    - G1 Z10 F500
    - G1 X150 Y150 F3600
    - G1 Z0 F500
    - ''
    customClass: btn
    name: Center
    offset: '2'
    width: '8'
  - commands:
    - G1 Z10 F500
    - G1 X10 F290 F3600
    - G1 Z0 F500
    customClass: btn
    name: Lower Left
    width: '2'
  - commands:
    - G1 Z10 F500
    - G1 X290 Y290 F3600
    - G1 Z0 F500
    customClass: btn
    name: Lower Right
    offset: '2'
    width: '2'
  customClass: ''
  layout: horizontal_grid
  name: CR-10 Bed Leveling
plugins:
  _disabled:
  - CR10_Leveling
  announcements:
    _config_version: 1
    channels:
      _blog:
        read_until: 1513605600
      _important:
        read_until: 1509547500
      _octopi:
        read_until: 1499253000
      _plugins:
        read_until: 1519603200
      _releases:
        read_until: 1513079100
  cura:
    cura_engine: /usr/local/bin/cura_engine
  discovery:
    publicPort: 80
    upnpUuid: 0a4d8722-7f6a-4f9b-9764-0d20c763f504
  softwareupdate:
    _config_version: 6
    check_providers:
      CR10_Leveling: CR10_Leveling
      fullscreen: fullscreen
    checks:
      octoprint:
        update_folder: /home/pi/OctoPrint
printerProfiles:
  default: _default
serial:
  autoconnect: true
  baudrate: 0
  port: AUTO
server:
  commands:
    serverRestartCommand: sudo service octoprint restart
    systemRestartCommand: sudo shutdown -r now
    systemShutdownCommand: sudo shutdown -h now
  firstRun: false
  onlineCheck:
    enabled: true
  pluginBlacklist:
    enabled: true
  secretKey: XzoyjZ5slibndRBLWQHey5pPekgmbhx0
  seenWizards:
    corewizard: 3
    cura: null
temperature:
  profiles:
  - bed: 100
    extruder: 210
    name: ABS
  - bed: 60
    extruder: 180
    name: PLA
  - bed: '60'
    extruder: '240'
    name: PETG
  - bed: '110'
    extruder: '250'
    name: ASA
webcam:
  ffmpeg: /usr/bin/avconv
  snapshot: http://127.0.0.1:8080/?action=snapshot
  stream: /webcam/?action=stream
  timelapse:
    fps: 25
    options:
      minDelay: 2.0
      retractionZHop: 0.0
    postRoll: 0
    type: zchange
```

electr0sheep, 2018-02-28:
@ryanneve It looks like you don't have any other plugins modifying controls, so you can remove the entire controls block, i.e.:

```
accessControl:
  salt: LCx0ofT7Jjq3co89ki2t7AeO2AZxO045
api:
  key: B13DEBC3EDA649ED8CBF6E5696003E82
plugins:
  _disabled:
  - CR10_Leveling
  announcements:
    _config_version: 1
    channels:
      _blog:
        read_until: 1513605600
      _important:
        read_until: 1509547500
      _octopi:
        read_until: 1499253000
      _plugins:
        read_until: 1519603200
      _releases:
        read_until: 1513079100
  cura:
    cura_engine: /usr/local/bin/cura_engine
  discovery:
    publicPort: 80
    upnpUuid: 0a4d8722-7f6a-4f9b-9764-0d20c763f504
  softwareupdate:
    _config_version: 6
    check_providers:
      CR10_Leveling: CR10_Leveling
      fullscreen: fullscreen
    checks:
      octoprint:
        update_folder: /home/pi/OctoPrint
printerProfiles:
  default: _default
serial:
  autoconnect: true
  baudrate: 0
  port: AUTO
server:
  commands:
    serverRestartCommand: sudo service octoprint restart
    systemRestartCommand: sudo shutdown -r now
    systemShutdownCommand: sudo shutdown -h now
  firstRun: false
  onlineCheck:
    enabled: true
  pluginBlacklist:
    enabled: true
  secretKey: XzoyjZ5slibndRBLWQHey5pPekgmbhx0
  seenWizards:
    corewizard: 3
    cura: null
temperature:
  profiles:
  - bed: 100
    extruder: 210
    name: ABS
  - bed: 60
    extruder: 180
    name: PLA
  - bed: '60'
    extruder: '240'
    name: PETG
  - bed: '110'
    extruder: '250'
    name: ASA
webcam:
  ffmpeg: /usr/bin/avconv
  snapshot: http://127.0.0.1:8080/?action=snapshot
  stream: /webcam/?action=stream
  timelapse:
    fps: 25
    options:
      minDelay: 2.0
      retractionZHop: 0.0
    postRoll: 0
    type: zchange
```

weinmannd, 2018-03-01:
also vote for an easier uninstall, my entire server got very unresponsive after installing the latest update

electr0sheep, 2018-03-02:
UPDATE: The latest version 2.0.0 should now check for this old legacy crap and get rid of it. So, if you are trying to get rid of the plugin, update to 2.0.0, and then you shouldn't have any issues. If you do, please let me know.

electr0sheep, 2018-03-13:
Since I haven't heard anything about this for a while, I'm closing the issue.

closed 2018-03-13 by electr0sheep