lsellens/octoprint-manual_bed_tramming #9: basic extensibility

BETLOG, 2018-02-27:
This would be MUCH more useful if the XY values could be user defined, and just call it a level assist plugin.

govido, 2018-02-27:
Why not use this instead?
https://plugins.octoprint.org/plugins/touchtest/

electr0sheep, 2018-02-28:
I have nearly finished work on this with the [customizable-controls branch](https://github.com/electr0sheep/OctoPrint-Cr10_leveling/tree/customizable-commands), but after saving the new settings, the user has to refresh the page for the changes to actually take effect. I think the solution might have something to do with the function rerenderControls() (I tried to find documentation for this function, but I couldn't)

electr0sheep, 2018-02-28:
https://github.com/electr0sheep/OctoPrint-Cr10_leveling/issues/4

electr0sheep, 2018-02-28:
#4

electr0sheep, 2018-02-28:
#4

BETLOG, 2018-02-28:
@govido Also good. But the point is that the bed parameters should either be read from the active octo profile, or user editable in the GUI.
Right now I'm using the 'custom control editor' to do exactly that. 
[image]

This (and the one you linked)  kind of 'convenience feature' plugin needs to have the same functionality, but in a no-brainer package that doesn't incorrectly assume it knows the bed size.

electr0sheep, 2018-03-02:
I think at this point, I have added some basic extensibility features, therefore I'm going to close this issue.

closed 2018-03-02 by electr0sheep