lsellens/octoprint-manual_bed_tramming #14: Is there way we could get an option to level for the S4 & S5?

devildog125, 2018-03-04:
Is there way we could get an option to level for the S4 & S5? The plugin works well however

electr0sheep, 2018-03-05:
Can you describe what you are wanting in more detail? In it's current state, the plugin should allow you to specify the coordinates of the 5 different positions. Are you requesting a drop-down menu of pre-configured defaults or something?

devildog125, 2018-03-05:
Correct, say a drop down for like CR-10, CR-10 S4 and CR-10 S5. Right now I have an S4 and when I use the the plugin, it thinks its a standard CR-10 and doesn't utilize the full bed. It just levels the area of 300 x 300 (the area of a standard CR-10). It needs to utilize the full 400 x 400 or 500 x 500 for an S5.

electr0sheep, 2018-03-12:
I have begun preliminary development on adding a pre-configured drop-down to make things easier, but it looks like it's going to be a pain in the butt to implement, so don't expect anything soon.

CptFredSureShot, 2018-03-14:
Hi there, and if you would like to include the mini as well, we would then have the full line-up!

bcsteeve, 2018-03-15:
But... why?  As @electr0sheep says, you can configure it in the options to whatever you want.

I was going to suggest this plugin drop the "CR-10" moniker entirely because there's really no reason - that I immediately can think of, anyway  - why it can't apply to any printer?

electr0sheep, 2018-03-23:
@bcsteeve Yeah you're right. At this point the plugin pretty much has completely diverged from what I was expecting it to be lol. I probably should follow your suggestion. Maybe if I have some extra time over the weekend or something I can look into it.

I think the main problem will be if I change the name, I'll need to make sure that plays nicely with the official plugin repo for updates, etc.

electr0sheep, 2018-04-20:
I'm going to close this issue.

The fix for this will be to get rid of CR-10 references, as the plugin is now pretty much fully customizable.

https://github.com/electr0sheep/OctoPrint-Cr10_leveling/issues/20

closed 2018-04-20 by electr0sheep