lsellens/octoprint-manual_bed_tramming #15: Suggestion: Option to switch front and read button position or edit labels

Havenwar, 2018-03-10:
I don't know if I'm just exceptionally prone to clicking things when I'm tired, but for some reason it just offends my spatial sense that the "front" buttons are above and the "rear" buttons are below, since when I sit at my printer and work on leveling the bed the front corners are "below" the rear corners from my point of view. 

Like it's not a big deal, but I just keep clicking the wrong buttons. I would consider switching the coordinates in the settings but since I can't change the labels I'd be stuck making the opposite mistake when for some reason I read the text instead. :D

So either an option to switch the position to match the physical layout of the corners, or maybe an option to change the label of the corners and center button and I can work it out myself, would be awesome.

marchyman, 2018-03-11:
Ditto, many times over.   I, too, would like the option of having the buttons swapped top to bottom so the location of a button matches the corner on the bed.   It also slightly annoys me that the buttons for the right are larger than the buttons for the left.

electr0sheep, 2018-03-12:
Regarding the buttons on the right being larger than the buttons on the left, I totally agree, and have tried to fix this by adding some extra spaces in between the words, i.e. 'top    left', but the extra whitespace didn't seem to take.

I'm normally pretty OCD about everything, and I guess I may have just never noticed the top and bottom buttons being in the wrong place, but now that you mention it, it's going to bug the fire out of me, so I'll swap the top and bottom rows in the next version.

electr0sheep, 2018-03-13:
It didn't end up being an issue with Atom, however I was able to resolve it anyway with 235cc430965074325e5de942092993d2f3f648dc

closed 2018-03-13 by electr0sheep