lsellens/octoprint-manual_bed_tramming #16: Suggestion: Option to set a custom Z

Havenwar, 2018-03-10:
Not everyone levels their bed with z at zero, so having the option to set a custom z height in addition to the existing options for x and y would be awesome so people have more options to do things their way.

electr0sheep, 2018-03-13:
@Havenwar I have implemented both of your suggestions, but I'm in the middle of a print so I can't test my Z axis changes right now. I will test and push a new version of the plugin tomorrow.

Havenwar, 2018-03-13:
Thanks! As for why the general idea I believe is that it's easier to feel the drag on say a 4mm drill bit than a 0.02mm feeler gauge, which makes it easier to get a very accurate and repeatable level. This also lets you level with a z0 of "true 0" instead of "0+thickness of a piece of paper" since you set the z to measure at to the exact thickness of the implement used to measure with. 

I am still experimenting with it so I couldn't say if it's better or not, but it's always nice to have options.

electr0sheep, 2018-03-13:
Just pushed 2.0.1 which contains the new settings

closed 2018-03-13 by electr0sheep