lsellens/octoprint-manual_bed_tramming #17: Better plugin description

ryanneve, 2018-03-14:
Just a small thing, but from the current description it's not entirely clear what this plugin does.  Perhaps replace:
## This plugins adds leveling controls for your CR-10 to the Control tab.
with something like:
## This plugin adds buttons which with one click position the print head over a corner leveling screw or the center of the print area.

My first thought was it had something to do with auto or mesh leveling.

bcsteeve, 2018-03-15:
Just to provide a counter-view...  your presumption that it has to do with auto or mesh leveling is your error in reading what isn't there, rather than misunderstanding what is there.  The description makes no mention of either auto or mesh or really says anything to suggest it could have anything to do with either.  (In fact, auto/mesh leveling systems have nothing to do with leveling... but that's a different discussion).

However, there's also no harm in being more specific about the description and I agree with ryanneve's wording, with a slight change (drop "one click" or move it to the end of sentence) for readability sake:

**This plugin adds buttons which position the print head over a corner leveling screw, or the center of the print area [, with one click]**

ryanneve, 2018-03-15:
Bcsteeve's suggestion looks great to me.

electr0sheep, 2018-03-19:
I updated the plugin repo. If you guys get a chance, let me know what you think.

ryanneve, 2018-03-21:
The README.md still just says "This plugins adds leveling controls for your CR-10 to the Control tab."
Perhaps I'm looking in the wrong place?

electr0sheep, 2018-03-22:
@ryanneve Oh, I thought you were talking about the description in the official plugin repo. I updated readme.md after realizing I was mistaken about what you meant.

closed 2018-03-22 by ryanneve