lsellens/octoprint-manual_bed_tramming #18: Lower Z coordinate

zoolanders49, 2018-04-02:
Hello,
could you add the ability to enter a decimal value for "Lower Z coordinate" .

Thank for your job. ;-)

PS:  I think you should rename your plugin with a more generic name because it is not specific to the CR-10 as you have made it fully customizable :)

electr0sheep, 2018-04-20:
I believe you should already be able to enter a decimal value for all the coordinates. The box will pretend like you can't do that, but you can still hit save all the same.

Nevertheless, I will specify a step of 0.1 so that that you won't run into this issue

Regarding the generic name for the plugin, I totally agree. The problem is, I would have to update the official plugin repo, and I'm kind of dreading that haha.

electr0sheep, 2018-06-12:
2.1.1 should take care of this.

closed 2018-06-12 by electr0sheep