lsellens/octoprint-manual_bed_tramming #20: Rename plugin to reflect the fact that it's generic now.

electr0sheep, 2018-04-20:
https://plugins.octoprint.org/help/registering/

AvanOsch, 2018-05-30:
I'm using this plugin for my Zonestar (Z8XM2).
Wouldn't have installed this if the plugin name had another printer name in it, but my printer is basically a Chinese rip-off of the CR-10 ;)

I edited the file
`/home/pi/oprint/python2.7/site-packages/octoprint_CR10_Leveling/static/cr10leveling.js`
And simply adjusted the "baseSettings" --> "name" to read "Bed Leveling"

Maybe adjust this in the plugin, and add some text to the description so people will know this works with other (most) printers as well...
I think you should even be able to change the plugin name, as long as the "plugin_identifier" remains the same...

electr0sheep, 2018-06-12:
> I think you should even be able to change the plugin name, as long as the "plugin_identifier" remains the same...
Oh I think you're right, I guess I won't have to submit a PR to the repo!