lsellens/octoprint-manual_bed_tramming #21: Z doesn't change anymore

johnnygizmo, 2018-05-01:
Since a recent update, the z doesn't change between button presses. Lower z is set to 0 and upper is set to 15.

johnnygizmo, 2018-05-02:
More info: On clicking a corner button, these are the only commands sent to the printer

Send: G90
Recv: ok
Send: G0 X270Y270F3600
Recv: ok

electr0sheep, 2018-06-12:
@johnnygizmo I just tested this with my CR-10, and the z was changing. For example, when I clicked `Bottom Left`, the z went up by 10 units, moved into the bottom left position, and the z went down by 10 units.

Can you confirm that I am understanding correctly, and if I am, what 3D printer are you using?

Can anyone else reproduce this issue?

johnnygizmo, 2018-06-12:
I reinstalled my Octopi (for other reasons) and the plugin works fine now. Sorry for the confusion.

closed 2021-04-23 by lsellens