lsellens/octoprint-manual_bed_tramming #27: Python 3 Compatibility

cp2004, 2020-06-27:
Hi there! I don't use this plugin (Don't have a CR10, but **at least 1,300 users do - It's popular**) but I am helping out plugins getting to Python 3 compatibility. 
As you are probably aware, OctoPrint currently supports Python 2 & 3 simultaneously. Users will begin migrating to Python 3 soon and your plugin will not be installable. As of March 2021, it will switch to Python 3 only.
Please see the guide [here](https://docs.octoprint.org/en/master/plugins/python3_migration.html)

Having taken a look at your plugin, there is not too much Python so if you don't want to update it, I could PR to sort it out. It could just need the compatibility flag detailed in the link above.

lsellens, 2020-10-28:
I've tested and this addon is python 3 compatible. It just needs the flag added.

ScruffR, 2020-11-06:
> I've tested and this addon is python 3 compatible. It just needs the flag added.

How did you install it? 
I've tried on my OctoPi but the plugin will never get enabled.

lsellens, 2020-11-08:
@ScruffR as cp2004 said you can edit the addon to include the missing tag or config octoprint to to force its use as described here
https://docs.octoprint.org/en/master/plugins/python3_migration.html#telling-octoprint-your-plugin-is-python-3-ready

ScruffR, 2020-11-08:
Yup, I tried @cp2004's version and it worked :+1:

PR #29 (2021-04-15): Add Python 3 compatibility string for Py2 & 3
Since Python 2 is EOL and all new users very shortly will be Py3, they would not be able to install and run this plugin.

There's the documentation here, if you want to setup dual environments for testing:
https://docs.octoprint.org/en/master/plugins/python3_migration.html

Or, to update your existing environment, one time script:
https://octoprint.org/blog/2020/09/10/upgrade-to-py3/

Closes #27 
Closes #28

Until this PR gets merged, and you (as a user) want to install the plugin - use this URL in the plugin manager > Get More> ...from URL:
```
ht [... cut]
commit 2819b260 (2021-04-15): Add Python 3 compatibility string for Py3 & 3

Since Python 2 is EOL and all new users very shortly will be Py3, they would not be able to install and run this plugin.

There's the documentation here, if you want to setup dual environments for testing:
https://docs.octoprint.org/en/master/plugins/python3_migration.html

Or, to update your existing environment, one time script:
https://octoprint.org/blog/2020/09/10/upgrade-to-py3/

Closes #27 
Closes #28
closed 2021-04-15 by electr0sheep