lsellens/octoprint-manual_bed_tramming #34: The Update

varek3d, 2021-11-06:
It's strange, I updated the OS on Octoprint.  After I see the buttons to help level my bed but they no longer do anything.  I get no control at all.  This is very disappointing I"ve been using the plug for years and I love it.  Any suggestions to repair it?

Thank you!

Sincerely,

B

Arturis, 2021-11-08:
I am also experiencing this same issue, the buttons are no longer functional after updating Octoprint today to 1.7.2.

theos10, 2021-11-21:
Same probleme, see log when clic on button : 
2021-11-21 19:02:30,119 - octoprint.util.comm - ERROR - Error while processing hook dashboard for phase sending and command G0 Z0F500:
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/util/comm.py", line 4611, in _process_command_phase
    tags=tags,
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/util/__init__.py", line 1737, in wrapper
    return f(*args, **kwargs)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_dashboard/__init__.py", line 811, in process_gcode
    CmdDict = dict ((x,float(y)) for d,x,y in (re.split('([A-Z])', i) for i in cmd.upper().split()))
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_dashboard/__init__.py", line 811, in <genexpr>
    CmdDict = dict ((x,float(y)) for d,x,y in (re.split('([A-Z])', i) for i in cmd.upper().split()))
ValueError: too many values to unpack (expected 3)

Thank for your help

rpoland007, 2021-11-30:
I get the same issue, the buttons are there, but they don't do anything.

varek3d, 2021-11-30:
Is there anything we can do other than not update?

nilshomeier, 2021-12-04:
Same for me. I noticed in the terminal window it says G90, when any of the position buttons are pressed. Heat and Stop Heating also work, but the movement buttons only send a G90.

varek3d, 2021-12-05:
is the developer going to help?

nilshomeier, 2021-12-05:
Ok, I figured out, what the issue is. Basically the buttons send gcode commands to the printer. In the current release the formatting of these commands is wrong. It sends "G0 Z10F500" when it should be "G0 Z10 F500". 

I fixed this in my installation, which was pretty simple actually.

ssh into your pi. navigate to the script folder: "cd oprint/lib/python3.7/site-packages/octoprint_CR10_Leveling/static"
and edit the script with "sudo nano cr10leveling.js"
[image]
You should see something like this, which is the javascript code for the buttons.
[image]
change the code by adding the space character before the F500 and Y like in the screenshot.

Save the file and restart octoprint, then the buttons should work again. At least they did for me.

theos10, 2021-12-05:
Hi,
Same probleme after change :
2021-12-05 15:12:16,987 - octoprint.util.comm - ERROR - Error while processing hook dashboard for phase sending and command G0 X40Y40F3600:
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/util/comm.py", line 4611, in _process_command_phase
    tags=tags,
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/util/__init__.py", line 1737, in wrapper
    return f(*args, **kwargs)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_dashboard/__init__.py", line 811, in process_gcode
    CmdDict = dict ((x,float(y)) for d,x,y in (re.split('([A-Z])', i) for i in cmd.upper().split()))
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_dashboard/__init__.py", line 811, in <genexpr>
    CmdDict = dict ((x,float(y)) for d,x,y in (re.split('([A-Z])', i) for i in cmd.upper().split()))
ValueError: too many values to unpack (expected 3)

There is a conflict with Dashborad plugin, it work fine when i disable DashBoard plugin.

AvanOsch, 2021-12-05:
I added spaces to the commands in my fork.
If someone would be willing to test it out:

https://github.com/AvanOsch/OctoPrint-Cr10_leveling

pilot76103, 2021-12-09:
> I added spaces to the commands in my fork. If someone would be willing to test it out:
> 
> https://github.com/AvanOsch/OctoPrint-Cr10_leveling
> 
> (I already created a pull request)

Just installed and tested, all appears well. Might want to update your readme to reference your link of 'https://github.com/AvanOsch/OctoPrint-Cr10_leveling/archive/master.zip'.

pilot76103, 2021-12-10:
@AvanOsch, I just wanted to come back and personally say thank you. This appears to have also fixed the responsiveness of my OctoPrint install. I kept having severe dropouts where it seemed like the interface was unresponsive and now it appears much better!

AvanOsch, 2021-12-10:
@pilot76103 You're very welcome.
Now let's hope @lsellens gets the main repo (and OctoPrint's update system) updated a.s.a.p....
The reason I don't have my fork updated with any info, is because I use it solely to push updates to the main branch.

lsellens, 2021-12-10:
Sorry first time I'm seeing this. I'll get it updated when I get home from work tonight.

lsellens, 2021-12-10:
Looks like my notifications got turned off for this repo some how. I have them on now, so I should be more responsive in the future.

AvanOsch, 2021-12-10:
@lsellens Cheers!

closed 2021-12-13 by lsellens