lsellens/octoprint-manual_bed_tramming #36: Settings do not change the configs.

leoheck, 2023-10-29:
Hey, I like your plugin but I am trying to change the Center but it does nothing.
Algo, it would be nice to get the boundaries from the Printer Profile as default instead of using 270 which is way out of the bad.

leoheck, 2023-10-29:
Do I have to restart Octoprint after changing the settings?

I am using Octoiprint 1.9.3

leoheck, 2023-10-29:
Oh, after restarting Octoprint it does work. When I save the settings it takes some seconds, I thought it was already appling the configs.

lsellens, 2023-11-02:
Yes you have to restart octoprint after settings change

leoheck, 2023-11-02:
Yeah, saw it later. There is a note written in the front page. I am not sure if I saw the same not in the settings page. It would make more sense there. Anyways, thanks for the replying.

closed 2023-11-02 by leoheck