lsellens/octoprint-manual_bed_tramming #37: No longer showing up on CONTROL page.

BATMan-Phln, 2025-05-23:
I've used this for several years.  I don't print much but a couple weeks ago I noticed that my leveling controls were gone.  Later I got a notification that my version of Python was out of date.  Yesterday I did a full, new install of OctoPrint.  Today I've been updating my plugins.  I installed the Bed Leveling Plugin and restarted the Pi.  It's still not showing up in the CONTROL page.  It's not disabled.  In fact, I did disable it, restarted the Pi, enabled it again, restarted and still no show.  

Any ideas?

Thanks,
Chad

[image]

lsellens, 2025-05-23:
Can you do me a favor and check if the issue was present in 2.1.4?

BATMan-Phln, 2025-05-23:
I'm not sure of the previous version that I had but, yes.  Like I said, it was doing the same thing for a week or so before I did the reinstall.  Even though I got a notification about Python, I wasn't in a hurry to redo everything. This was the only plugin that I really wanted to work.  That's the main reason I did the reinstall.

lsellens, 2025-05-23:
I'll check it out when I get home tonight. If you need it up in running right away just install 2.1.3 that should definitely work.

lsellens, 2025-05-23:
Thanks so much for checking that it saves me some work. I'm pretty sure I know what the issue is already, but I'll confirm when I get home in a few hours.

lsellens, 2025-05-24:
Should be fixed with 692a8b1133d74877fc114ab4524495852f5bdc61
I'll push a new version out

lsellens, 2025-05-24:
While testing this I ran into another bug, but I'm so tired I can hardly see straight right now. I'll have to get back to this another day.

closed 2025-05-24 by lsellens